pebble-dev/pebble-firmware #2: All of the system fonts

ethanc8, 2025-01-28:


Willow-Systems, 2025-01-28:
We can use https://renaissance.rebble.io/

ethanc8, 2025-01-28:
Raster Gothic can probably be replaced by the Rebble project's [Renaissance font](https://github.com/pebble-dev/renaissance), which is metric-compatible. Here are [the other system fonts](https://developer.rebble.io/developer.pebble.com/guides/app-resources/system-fonts/index.html) that we'd need to replace:

* Bitham (rasterization of Gotham, we should probably rasterize one of the [Google Fonts geometric fonts](https://fonts.google.com/?script=Latn&categoryFilters=Sans+Serif:%2FSans%2FGeometric))
* Roboto, Roboto Bold, Droid Serif (we should probably rasterize newer versions of Roboto and Noto Serif)
* LECO (idk about this)
* Emojis in Raster Gothic

If we only cared about US copyright law, we could probably just copy the bitmaps from the binary Pebble firmware (US copyright law only covers splines and other programs that produce fonts (OTF and TTF files are considered programs), not the font designs themselves).

jwise, 2025-01-29:
Yes, we could certainly provide a script so that users could extract their own fonts from a firmware image that they happen to have.

jplexer, 2025-01-31:
working with the code I know that LECO and Droid Serif are already included so we can already check those

jneubrand, 2025-01-31:
[LECO 1976 Regular](https://www.myfonts.com/products/regular-leco-1976-272584) seems to be the source for the `LECO_BOLD` fonts and is available for free officially, though the rest of the family (notably light) isn’t. Also, the license doesn’t seem to be compatible with OS inclusion. One could just try emailing the [creators](https://www.carnokytype.com/) to ask if we can use those two weights’ number glyphs with attribution :)

Hexxeh, 2025-02-12:
Most of these have been added as binaries on the coredevices fork which should be getting moved across here soon:

https://github.com/coredevices/pebble/commit/1a7d6575c1291bc589e6f8f43fd8bc3b691bb687
https://github.com/coredevices/pebble/commit/ba92694e2990820f38c35dea4242ab7fa67e1def
https://github.com/coredevices/pebble/commit/e9c52bf5e7595e59896a6f19a29fc5d7772eb8bd
https://github.com/coredevices/pebble/commit/bcf229f3c19b6ad8b2dea4da3402de4d75f1952a

Hexxeh, 2025-02-13:
The remaining fonts (AgencyFB) have been added, so this should be fixed now.

closed 2025-02-13 by Hexxeh

waldyrious, 2025-03-18:
> The remaining fonts (AgencyFB) have been added

For reference, that was done in https://github.com/pebble-dev/pebble-firmware/pull/25.