pebble-dev/pebble-firmware #14: Get snowy-qemu booting again

jwise, 2025-02-10:
Now that we have PebbleOS building for nRF hardware, we probably need to un-mothball STM32 hardware at least enough to build and boot in snowy-qemu, so that people can begin hacking on PebbleOS on their desk.

AC: at least one app can be installed in snowy-qemu emulator through classic libpebble2 emulation environment

Hexxeh, 2025-02-13:
This should be working now. I haven't tested appinstall because I don't have libpebble setup but it boots and GDB/pulse console works.

Hexxeh, 2025-02-18:
Tested with rebble-tool (https://github.com/richinfante/rebbletool) and was able to install to QEMU, though you need to make sure there's a tag for the commit you're building from or below else libpebble blows up on the missing version info. We should push a `v4.3` tag to the repo to fix this (last version on the internal repo).

closed 2025-02-18 by Hexxeh