pebble-dev/pebble-firmware #48: Add BetterNotification concept

jplexer, 2025-02-20:
BetterNotifications is a design concept by @lavglaab that I really think we should implement (in fact I already started an implementation).

Just opening this issue to track/collect thoughts!

https://www.figma.com/design/qlKMznYUgqes8dhsKKg9Ds/Notifications

lavglaab, 2025-03-04:
See PR #52 -
AttributeIDBannerText for notifications is done on snowy, spalding, and silk. When a notification has associated BannerText it will show as appropriate for the platform, and it falls back gracefully to the current design if no BannerText is present. 

Also, silk now uses inverted white icons on a black header, rather than a dithered grey header.

andiohn, 2025-03-22:
Looks great!

tilden, 2025-03-24:
I think this could make a nice option for users to have, but I'd be concerned about making it the default.

IMO, the icons alone should be sufficient to convey from which app the notification originated. (Though I realize right now there are a number of now-common apps with missing icons.)