pebble-dev/pebble-firmware #57: building: Could not find the program ['clang']

benjamin051000, 2025-03-01:
https://github.com/pebble-dev/pebble-firmware/blob/a200e39f7e7de113d7f9a9557f6204d292414e72/README.md?plain=1#L74

I see the following:
```
❯ ./waf configure --board silk_bb2 --nojs
Setting top to                           : /home/benjamin/src/pebble/pebble-firmware 
Setting out to                           : /home/benjamin/src/pebble/pebble-firmware/build 
Checking for program 'xgettext'          : /usr/bin/xgettext 
Checking for program 'msgcat'            : /usr/bin/msgcat 
Checking for program 'node, nodejs'      : /usr/bin/node 
Checking for program 'nanopb_generator'  : /home/benjamin/src/pebble/pebble-firmware/.venv/bin/nanopb_generator 
reset_config: trst_and_srst srst_nogate
Configuring ARM cortex-m0 environment 
Using Bluetooth controller: da14681-01 
Configuring arm_firmware environment 
Debug Mode
Configuring unit test environment 
Could not find the program ['clang']
(complete log in /home/benjamin/src/pebble/pebble-firmware/build/config.log)
```

Is clang a build dependency? If so, it should be specified. I have `arm-none-eabi-*` installed on my PATH, and assumed that would be sufficient... what is clang needed for?

Hexxeh, 2025-03-01:
Clang is required for building binaries that run on the host.

benjamin051000, 2025-03-01:
Okay, gotcha. I'll add this to the README

gmarull, 2025-04-15:
https://github.com/pebble-dev/pebble-firmware/pull/154

closed 2025-04-15 by gmarull