pebble-dev/pebble-firmware #59: `tools/generate_native_sdk/parse_c_decl.py` can't find `stdbool.h`

benjamin051000, 2025-03-01:
I'm seeing the following build error.

# Setup
```
./waf configure --board snowy_bb2 --qemu --nojs
./waf build
```
The output looks like this:
```
Waf: Entering directory `/home/benjamin/src/pebble/pebble-firmware/build'
Could not find gitignore, but it's defined in templates.json
get_git_version: unable to determine git revision
get_git_revision: Invalid git tag! Must follow this form: `v0[.0[.0]][-suffix]`
[   1/1565] Processing src/idl/nanopb/simple.proto
[   2/1565] Processing src/idl/nanopb/event.proto
[   3/1565] Processing src/idl/nanopb/activity.proto

# ...lots more output, omitted for brevity...

[ 657/1565] Compiling src/fw/services/normal/analytics/analytics_external.c
WARNING:root:Omitting 'UUID_INVALID' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'BLE Bluetooth Low Energy' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'GRangeHorizontal' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'GRangeVertical' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'graphics_text_perimeter_debugging_enable' from SDK export because its revision (999) is higher than the current revision (89)
Traceback (most recent call last):
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/generate_pebble_native_sdk_files.py", line 186, in <module>
    generate_shim_files(
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/generate_pebble_native_sdk_files.py", line 97, in generate_shim_files
    extract_symbol_info(files, functions, types, defines, pbl_output_src_dir,
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/extract_symbol_info.py", line 78, in extract_symbol_info
    parse_c_decl.parse_file(all_headers_file, filenames,
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/parse_c_decl.py", line 284, in parse_file
    raise ParsingException(error_str)
generate_native_sdk.parse_c_decl.parse_file.<locals>.ParsingException: Fatal: <Diagnostic severity 4, location <SourceLocation file '/home/benjamin/src/pebble/pebble-firmware/src/fw/drivers/ambient_light.h', line 19, column 10>, spelling "'stdbool.h' file not found">
```

---

I tested this in a separate project just to make sure my installation of the arm toolchain was correct:
```
~/src/test-armgcc via C v14.2.1-gcc 
❯ cat main.c
#include <stdbool.h>

~/src/test-armgcc via C v14.2.1-gcc 
❯ arm-none-eabi-gcc -E main.c 
# 0 "main.c"
# 0 "<built-in>"
# 0 "<command-line>"
# 1 "main.c"
# 1 "/home/benjamin/.utils/lib/gcc/arm-none-eabi/14.2.1/include/stdbool.h" 1 3 4
# 2 "main.c" 2
```

As you can see, I do in fact have `stdbool.h` and `arm-none-eabi-gcc` can see it in this context.

Any ideas on why the python tool isn't able to see this? 

---
Output of `./waf build -v` (the last few lines):
```
[ 665/1565] Compiling src/fw/services/normal/voice_endpoint.c
10:24:24 runner ['arm-none-eabi-gcc', '-std=c11', '-I/home/benjamin/src/pebble/pebble-firmware/src/fw/util/time', '-Wall', '-Wextra', '-Werror', '-Wpointer-arith', '-Wno-unused-parameter', '-Wno-missing-field-initializers', '-Wno-error=unused-function', '-Wno-error=unused-variable', '-Wno-error=unused-parameter', '-Wno-error=unused-const-variable', '-Wno-error=unused-but-set-variable', '-Wno-packed-bitfield-compat', '-Wno-address-of-packed-member', '-Wno-enum-int-mismatch', '-Wno-expansion-to-defined', '-Wno-enum-conversion', '-fvar-tracking-assignments', '-mthumb', '-ffreestanding', '-ffunction-sections', '-fbuiltin', '-fno-builtin-itoa', '-g3', '-gdwarf-4', '-flto', '-flto-partition=balanced', '--param', 'lto-partitions=128', '-fuse-linker-plugin', '-fno-if-conversion', '-fno-caller-saves', '-fira-region=mixed', '-finline-functions', '-fconserve-stack', '--param', 'inline-unit-growth=1', '--param', 'max-inline-insns-auto=1', '--param', 'max-cse-path-length=1000', '--param', 'max-grow-copy-bb-insns=1', '-fno-hoist-adjacent-loads', '-fno-optimize-sibling-calls', '-fno-schedule-insns2', '-mcpu=cortex-m4', '-Os', '-Isrc/fw', '-I../src/fw', '-Isrc/fw/applib/vendor/uPNG', '-I../src/fw/applib/vendor/uPNG', '-Isrc/fw/applib/vendor/tinflate', '-I../src/fw/applib/vendor/tinflate', '-Isrc/fw/vendor/CMSIS/CMSIS/Core/Include', '-I../src/fw/vendor/CMSIS/CMSIS/Core/Include', '-Isrc/fw/vendor/stm32-sdk/Device/ST/STM32F4xx/Include', '-I../src/fw/vendor/stm32-sdk/Device/ST/STM32F4xx/Include', '-Isrc/fw/vendor/stm32-sdk/STM32F4xx_StdPeriph_Driver/inc', '-I../src/fw/vendor/stm32-sdk/STM32F4xx_StdPeriph_Driver/inc', '-Isrc/libc/include', '-I../src/libc/include', '-Isrc/libbtutil/include', '-I../src/libbtutil/include', '-Isrc/libos/include', '-I../src/libos/include', '-Isrc/include', '-I../src/include', '-Isrc/libutil/includes', '-I../src/libutil/includes', '-Isrc/idl', '-Isrc/fw/vendor/nanopb', '-I../src/fw/vendor/nanopb', '-Isrc/fw/vendor/FreeRTOS/Source/include', '-I../src/fw/vendor/FreeRTOS/Source/include', '-Isrc/fw/vendor/FreeRTOS/Source/portable/GCC/ARM_CM4_PEBBLE', '-I../src/fw/vendor/FreeRTOS/Source/portable/GCC/ARM_CM4_PEBBLE', '-DSTM32F429_439xx', '-DMAX_FONT_GLYPH_SIZE=256', '-DTARGET_QEMU', '-DIS_BIGBOARD', '-DPBL_LOG_ENABLED', '-DPBL_LOGS_HASHED', '-DBOOTLOADER_TEST_STAGE1=0', '-DBOOTLOADER_TEST_STAGE2=0', '-DPULSE_EVERYWHERE=1', '-DBT_CONTROLLER_QEMU', '-DUSE_STDPERIPH_DRIVER=1', '-DBOARD_SNOWY_BB2=1', '-DPLATFORM_SNOWY=1', '-DMICRO_FAMILY_STM32F4=1', '-DSCREEN_COLOR_DEPTH_BITS=8', '-DMIN_SDK_VERSION=2', '-DCAPABILITY_HAS_CORE_NAVIGATION4=1', '-DCAPABILITY_HAS_MAGNETOMETER=1', '-DCAPABILITY_HAS_APPLE_MFI=1', '-DCAPABILITY_HAS_SDK_SHELL4=1', '-DCAPABILITY_HAS_VIBE_SCORES=1', '-DCAPABILITY_HAS_LAUNCHER4=1', '-DCAPABILITY_HAS_TIMELINE_PEEK=1', '-DCAPABILITY_HAS_MAPPABLE_FLASH=1', '-DCAPABILITY_HAS_TEMPERATURE=1', '-DCAPABILITY_HAS_HARDWARE_PANIC_SCREEN=1', '-DCAPABILITY_HAS_DEFECTIVE_FW_CRC=1', '-DCAPABILITY_COMPOSITOR_USES_DMA=1', '-DCAPABILITY_HAS_HEALTH_TRACKING=1', '-DCAPABILITY_HAS_WEATHER=1', '-DCAPABILITY_HAS_APP_GLANCES=1', '-DCAPABILITY_HAS_PMIC=1', '-DCAPABILITY_USE_PARALLEL_FLASH=1', '-DCAPABILITY_HAS_MASKING=1', '-DCAPABILITY_HAS_SPRF_V3=0', '-DCAPABILITY_HAS_LED=0', '-DCAPABILITY_HAS_PUTBYTES_PREACKING=0', '-DCAPABILITY_HAS_TOUCHSCREEN=0', '-DCAPABILITY_HAS_GLYPH_BITMAP_CACHING=0', '-DCAPABILITY_HAS_MICROPHONE=0', '-DCAPABILITY_HAS_ACCESSORY_CONNECTOR=0', '-DCAPABILITY_HAS_JAVASCRIPT=0', '-DCAPABILITY_HAS_BUILTIN_HRM=0', '-DSHOW_PEBBLE_JUST_RESET_ALERT', '-DSHOW_BAD_BT_STATE_ALERT', '-DGCC_ARM_CM4_PEBBLE', '../src/fw/services/normal/voice_endpoint.c', '-c', '-o/home/benjamin/src/pebble/pebble-firmware/build/src/fw/services/normal/voice_endpoint.c.2.o']
WARNING:root:Omitting 'UUID_INVALID' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'BLE Bluetooth Low Energy' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'GRangeHorizontal' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'GRangeVertical' from SDK export because its revision (999) is higher than the current revision (89)
WARNING:root:Omitting 'graphics_text_perimeter_debugging_enable' from SDK export because its revision (999) is higher than the current revision (89)
Traceback (most recent call last):
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/generate_pebble_native_sdk_files.py", line 186, in <module>
    generate_shim_files(
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/generate_pebble_native_sdk_files.py", line 97, in generate_shim_files
    extract_symbol_info(files, functions, types, defines, pbl_output_src_dir,
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/extract_symbol_info.py", line 78, in extract_symbol_info
    parse_c_decl.parse_file(all_headers_file, filenames,
  File "/home/benjamin/src/pebble/pebble-firmware/tools/generate_native_sdk/parse_c_decl.py", line 284, in parse_file
    raise ParsingException(error_str)
generate_native_sdk.parse_c_decl.parse_file.<locals>.ParsingException: Fatal: <Diagnostic severity 4, location <SourceLocation file '/home/benjamin/src/pebble/pebble-firmware/src/fw/drivers/ambient_light.h', line 19, column 10>, spelling "'stdbool.h' file not found">

Waf: Leaving directory `/home/benjamin/src/pebble/pebble-firmware/build'
Build failed
```

gmarull, 2025-03-01:
Where was toolchain installed from? What does version report?

benjamin051000, 2025-03-01:
My OS is OpenSUSE Tumbleweed.

I installed the arm toolchain via the readme by downloading the tarball, extracted it into a dir. Then I added the following to my zshrc:

```bash
UTILS="$HOME/.utils"

export PATH="$UTILS/bin:$PATH"
export MANPATH="$UTILS/share:$MANPATH"

# Set up C libraries and whatnot
export C_INCLUDE_PATH="$UTILS/include:$C_INCLUDE_PATH"
export CPLUS_INCLUDE_PATH="$UTILS/include:$CPLUS_INCLUDE_PATH"

export LIBRARY_PATH="$UTILS/lib:$LIBRARY_PATH"
export LD_LIBRARY_PATH="$UTILS/lib:$LD_LIBRARY_PATH"
```

benjamin051000, 2025-03-02:
It looks like I fixed this error. My `C_INCLUDE_PATH` was wrong, as turns out the arm toolchain install has multiple nested `include/` dirs.

```bash
export C_INCLUDE_PATH="$arm_path/lib/gcc/arm-none-eabi/14.2.1/include:$arm_path/include:$arm_path/arm-none-eabi/include:$C_INCLUDE_PATH"
```

This solves that error.

closed 2025-03-02 by benjamin051000