pebble-dev/pebble-firmware #77: Fix Unit Tests: test_app_fetch_endpoint.c

Colin-Suckow, 2025-03-07:
## Summary
A number of the unit tests fail to compile, preventing `waf test` from running. This task covers resolving the build issues in `test_app_fetch_endpoint.c`

## End State
`./waf test` should be able tun run the test suite when this test is removed from the `bld.env.BROKEN_TESTS` list.

waldyrious, 2025-03-18:
How does this issue differ from #46? And how does it relate to #69? (I think I could sort of piece the answers together from the discussion in #69, but I don't want to make assumptions.)

Colin-Suckow, 2025-03-19:
> How does this issue differ from [#46](https://github.com/pebble-dev/pebble-firmware/issues/46)? And how does it relate to [#69](https://github.com/pebble-dev/pebble-firmware/pull/69)? (I think I [...]

Good question. #46 is about `./waf test` failing to run at all. In #69 we fixed the issues stopping the test suite, but that exposed a number of tests that can't compile due to a variety of issues. Instead of ballooning #69 into a PR that tries to fix a bunch of different issues, we created this issue to cover fixing the specific broken tests.