pebble-dev/pebble-firmware #110: No Text Channels on Discord Link

joachimweyl, 2025-03-19:
This [link](https://discord.com/channels/221364737269694464/256091649996881922) on this [page](https://ericmigi.com/blog/introducing-two-new-pebbleos-watches) takes you to a discord page that says "NO TEXT CHANNELS"

joachimweyl, 2025-03-19:
where as this [link](http://discord.gg/aRUAYFN) takes you to The Rebble Alliance discord on this [page](https://github.com/pebble-dev/wiki/wiki).
Is this the correct discord for the discussion about the 2 new pebbles?
If so can the link in the issue description be updated to the link in this comment?

jwise, 2025-03-19:
not a PebbleOS issue, closing.  feel free to mail Eric if you have questions about his blog post

closed 2025-03-19 by jwise

ericmigi, 2025-03-19:
Thanks, I'll fix it but in future you can dm me on discord for this stuff