pebble-dev/pebble-firmware #122: Feature: "In memory" option to keep certain apps in memory

andiohn, 2025-03-22:
I believe this 'kinda' works, but it would be cool to be able to select 2-3 apps to stay on the watch. Some times this works and I'd love to keep like, music, authenticator, and stopwatch for instance.

jwise, 2025-03-23:
This is a task better suited for the mobile app, not the firmware, I believe.

jwise, 2025-03-23:
Ah, I was wrong.  I am told that the app cache is managed by the firmware.  It seems like something that would need help both from the mobile app and the firmware, though.  Reopening as a would-be-nice some day.

andiohn, 2025-03-29:
Ya, I believe there was a small hack on how to do it before by sideloading or something. I'll have to see if I can find it back in the past. My thought is this, you could set 'priority' on a couple apps, like 1, 2, 3, 4 or something and they would be removed from the watch if required by memory pressure in order. Usually people use something like 2-3 apps often, and suddenly they would just immediately open. With watches with larger memory, then they might never get removed. Also, perhaps the mobile app could check to see if they're still in memory every so often, and if they are no longer in memory, try to add one or something. 

I know the pebbles are pressed for memory, but if I remember correctly, I was able to keep about 3 before it would have to stream the 4th one.