pebble-dev/pebble-firmware #125: Add host-based address resolution support to Nimble

Hexxeh, 2025-03-27:
CC2564X doesn't support resolving RPAs on the controller, the `LE Set Address Resolution Enable (0x202d)` and `LE Add Device to Resolving List (0x2027)` commands return `Unknown HCI Command` and Nimble just ignores it.

We'll need to add support to Nimble to do resolution host side.

See:

https://github.com/apache/mynewt-nimble/issues/897
https://github.com/espressif/esp-nimble/commit/fdfa72957d5e26ede402e96f0e1e6b608bcd9986

jwise, 2025-03-28:
We could cherry pick this to a fork of our own that lives in pebble-dev that has otherwise-upstream NimBLE, I suppose.  Either that or we could agitate to upstream it, depending on how good of citizens we want to be...

Hexxeh, 2025-03-28:
That's what I'm thinking short/medium term. I don't know how upstreaming would work giving we'd be cherry-picking Espressif's changes, probably can't upstream them on their behalf.

gmarull, 2025-03-31:
I'd try to at least open the PR upstream when we have it ready on our fork. If Espressif posts content on Apache, I don't see any issues as soon as they're credited (maybe sync with them or let them know about our intention). What I'd like to avoid is to deviate too much from upstream NimBLE as we do not have the resources now to maintain a BLE stack fork. All the tiny issues I found, I opened a PR upstream (one was merged, other is pending).

Hexxeh, 2025-03-31:
After looking at it a bit more I decided not to base on Espressif's work directly, some of what they've done seems more complex than it feels like it needs to be.

I took a stab at doing host based resolution based on the identity address/IRKs we already store, but had problems matching the IRK to the RPA we get.

closed 2025-04-04 by Hexxeh