pebble-dev/pebble-firmware #129: Build errors on running macOS M1 Pro (Sonoma)

jibraniqbal666, 2025-04-01:
Hey team! 

I am trying to run this with qemu as i don't have an hardware and desired to run in vm.

These are my tools and errors respectively.

**TLDR! it is failing while building**

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GCC) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

```
$ python3 --version
Python 3.13.2
```

```
$ ./waf configure --board=snowy_bb2 --qemu --nojs

Setting top to                           : /Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp
Setting out to                           : /Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/build
Checking for program 'xgettext'          : /opt/homebrew/bin/xgettext
Checking for program 'msgcat'            : /opt/homebrew/bin/msgcat
Checking for program 'node, nodejs'      : /opt/homebrew/bin/node
Checking for program 'nanopb_generator'  : /Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/.venv/bin/nanopb_generator
reset_config: trst_and_srst srst_nogate
openocd is not Pebble flavored!
Log level: INFO
Sandbox disabled
Turning on LTO.
Configuring ARM cortex-m0 environment
Using Bluetooth controller: qemu
Configuring arm_firmware environment
Debug Mode
Configuring unit test environment
Configuring 32 bit host environment
Configuring stored apps environment
Checking requirements.txt
Checking requirements-brew.txt
'configure' finished successfully (2.038s)

```

```
./waf build -v
```

```
[1180/1555] Compiling src/fw/services/normal/music.c
22:18:21 runner ['arm-none-eabi-gcc', '-std=c11', '-I/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/src/fw/util/time', '-Wall', '-Wextra', '-Werror', '-Wpointer-arith', '-Wno-unused-parameter', '-Wno-missing-field-initializers', '-Wno-error=unused-function', '-Wno-error=unused-variable', '-Wno-error=unused-parameter', '-Wno-error=unused-const-variable', '-Wno-error=unused-but-set-variable', '-Wno-packed-bitfield-compat', '-Wno-address-of-packed-member', '-Wno-enum-int-mismatch', '-Wno-expansion-to-defined', '-Wno-enum-conversion', '-fvar-tracking-assignments', '-mthumb', '-ffreestanding', '-ffunction-sections', '-fbuiltin', '-fno-builtin-itoa', '-g3', '-gdwarf-4', '-flto', '-flto-partition=balanced', '--param', 'lto-partitions=128', '-fuse-linker-plugin', '-fno-if-conversion', '-fno-caller-saves', '-fira-region=mixed', '-finline-functions', '-fconserve-stack', '--param', 'inline-unit-growth=1', '--param', 'max-inline-insns-auto=1', '--param', 'max-cse-path-length=1000', '--param', 'max-grow-copy-bb-insns=1', '-fno-hoist-adjacent-loads', '-fno-optimize-sibling-calls', '-fno-schedule-insns2', '-mcpu=cortex-m4', '-Os', '-Isrc/fw', '-I../src/fw', '-Isrc/fw/applib/vendor/uPNG', '-I../src/fw/applib/vendor/uPNG', '-Isrc/fw/applib/vendor/tinflate', '-I../src/fw/applib/vendor/tinflate', '-Isrc/libbtutil/include', '-I../src/libbtutil/include', '-Isrc/libos/include', '-I../src/libos/include', '-Isrc/include', '-I../src/include', '-Isrc/libutil/includes', '-I../src/libutil/includes', '-Isrc/idl', '-Ithird_party/nanopb/nanopb', '-I../third_party/nanopb/nanopb', '-Ithird_party/hal_stm32/stm32-sdk/Device/ST/STM32F4xx/Include', '-I../third_party/hal_stm32/stm32-sdk/Device/ST/STM32F4xx/Include', '-Ithird_party/hal_stm32/stm32-sdk/STM32F4xx_StdPeriph_Driver/inc', '-I../third_party/hal_stm32/stm32-sdk/STM32F4xx_StdPeriph_Driver/inc', '-Ithird_party/cmsis_core/CMSIS/CMSIS/Core/Include', '-I../third_party/cmsis_core/CMSIS/CMSIS/Core/Include', '-Isrc/libc/include', '-I../src/libc/include', '-Ithird_party/freertos', '-I../third_party/freertos', '-Ithird_party/freertos/FreeRTOS-Kernel/FreeRTOS/Source/include', '-I../third_party/freertos/FreeRTOS-Kernel/FreeRTOS/Source/include', '-Ithird_party/freertos/FreeRTOS-Kernel/FreeRTOS/Source/portable/GCC/ARM_CM4F', '-I../third_party/freertos/FreeRTOS-Kernel/FreeRTOS/Source/portable/GCC/ARM_CM4F', '-DSTM32F429_439xx', '-DMAX_FONT_GLYPH_SIZE=256', '-DTARGET_QEMU', '-DIS_BIGBOARD', '-DDEFAULT_LOG_LEVEL=LOG_LEVEL_INFO', '-DPBL_LOG_ENABLED', '-DPBL_LOGS_HASHED', '-DBOOTLOADER_TEST_STAGE1=0', '-DBOOTLOADER_TEST_STAGE2=0', '-DPULSE_EVERYWHERE=1', '-DBT_CONTROLLER_QEMU', '-DUSE_STDPERIPH_DRIVER=1', '-DBOARD_SNOWY_BB2=1', '-DPLATFORM_SNOWY=1', '-DMICRO_FAMILY_STM32F4=1', '-DSCREEN_COLOR_DEPTH_BITS=8', '-DMIN_SDK_VERSION=2', '-DCAPABILITY_HAS_APP_GLANCES=1', '-DCAPABILITY_HAS_PMIC=1', '-DCAPABILITY_HAS_TEMPERATURE=1', '-DCAPABILITY_HAS_MAGNETOMETER=1', '-DCAPABILITY_HAS_WEATHER=1', '-DCAPABILITY_HAS_MAPPABLE_FLASH=1', '-DCAPABILITY_USE_PARALLEL_FLASH=1', '-DCAPABILITY_HAS_DEFECTIVE_FW_CRC=1', '-DCAPABILITY_HAS_CORE_NAVIGATION4=1', '-DCAPABILITY_HAS_TIMELINE_PEEK=1', '-DCAPABILITY_HAS_LAUNCHER4=1', '-DCAPABILITY_HAS_HEALTH_TRACKING=1', '-DCAPABILITY_HAS_HARDWARE_PANIC_SCREEN=1', '-DCAPABILITY_HAS_MASKING=1', '-DCAPABILITY_HAS_SDK_SHELL4=1', '-DCAPABILITY_COMPOSITOR_USES_DMA=1', '-DCAPABILITY_HAS_APPLE_MFI=1', '-DCAPABILITY_HAS_VIBE_SCORES=1', '-DCAPABILITY_HAS_BUILTIN_HRM=0', '-DCAPABILITY_HAS_PUTBYTES_PREACKING=0', '-DCAPABILITY_HAS_TOUCHSCREEN=0', '-DCAPABILITY_HAS_GLYPH_BITMAP_CACHING=0', '-DCAPABILITY_HAS_MICROPHONE=0', '-DCAPABILITY_HAS_SPRF_V3=0', '-DCAPABILITY_HAS_JAVASCRIPT=0', '-DCAPABILITY_HAS_ACCESSORY_CONNECTOR=0', '-DCAPABILITY_HAS_LED=0', '-DSHOW_PEBBLE_JUST_RESET_ALERT', '-DSHOW_BAD_BT_STATE_ALERT', '-DGCC_ARM_CM4F', '-DFREERTOS_USE_TIMERS=0', '../src/fw/services/normal/music.c', '-c', '-o/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/build/src/fw/services/normal/music.c.2.o']
../stored_apps/golf/src/golf.c:19:10: fatal error: sys/types.h: No such file or directory
   19 | #include <sys/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.

In file included from sdk/basalt/include/pebble_process_info.h:19,
                 from stored_apps/golf/appinfo.auto.c:1:
/opt/homebrew/Cellar/arm-none-eabi-gcc/14.2.0/lib/gcc/arm-none-eabi/14.2.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.

Waf: Leaving directory `/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/build'
Build failed
 -> task in '/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/build/stored_apps/golf/pebble-app.elf' failed with exit status 1:
        {task 4394063520: c golf.c -> golf.c.4.o}
['arm-none-eabi-gcc', '-std=c99', '-mcpu=cortex-m3', '-mthumb', '-ffunction-sections', '-fdata-sections', '-g', '-fPIE', '-Os', '-D_TIME_H_', '-Wall', '-Wextra', '-Werror', '-Wno-unused-parameter', '-Wno-error=unused-function', '-Wno-error=unused-variable', '-fPIE', '-DPBL_PLATFORM_BASALT', '-DPBL_COLOR', '-DPBL_RECT', '-DPBL_MICROPHONE', '-DPBL_SMARTSTRAP', '-DPBL_HEALTH', '-DPBL_COMPASS', '-DPBL_SMARTSTRAP_POWER', '-DPBL_DISPLAY_WIDTH=144', '-DPBL_DISPLAY_HEIGHT=168', '-Isdk/basalt/include', '-Istored_apps/golf', '-I../stored_apps/golf', '-Istored_apps/golf/src', '-I../stored_apps/golf/src', '-DSTM32F429_439xx', '-DMAX_FONT_GLYPH_SIZE=256', '-DTARGET_QEMU', '-DIS_BIGBOARD', '-DDEFAULT_LOG_LEVEL=LOG_LEVEL_INFO', '-DPBL_LOG_ENABLED', '-DPBL_LOGS_HASHED', '-DBOOTLOADER_TEST_STAGE1=0', '-DBOOTLOADER_TEST_STAGE2=0', '-DPULSE_EVERYWHERE=1', '-DBT_CONTROLLER_QEMU', '-DRELEASE', '../stored_apps/golf/src/golf.c', '-c', '-o/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/build/stored_apps/golf/src/golf.c.4.o']
 -> task in '/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/build/stored_apps/golf/pebble-app.elf' failed with exit status 1:
        {task 4394063408: c appinfo.auto.c -> appinfo.auto.c.4.o}
['arm-none-eabi-gcc', '-std=c99', '-mcpu=cortex-m3', '-mthumb', '-ffunction-sections', '-fdata-sections', '-g', '-fPIE', '-Os', '-D_TIME_H_', '-Wall', '-Wextra', '-Werror', '-Wno-unused-parameter', '-Wno-error=unused-function', '-Wno-error=unused-variable', '-fPIE', '-DPBL_PLATFORM_BASALT', '-DPBL_COLOR', '-DPBL_RECT', '-DPBL_MICROPHONE', '-DPBL_SMARTSTRAP', '-DPBL_HEALTH', '-DPBL_COMPASS', '-DPBL_SMARTSTRAP_POWER', '-DPBL_DISPLAY_WIDTH=144', '-DPBL_DISPLAY_HEIGHT=168', '-Isdk/basalt/include', '-Istored_apps/golf', '-I../stored_apps/golf', '-Istored_apps/golf/src', '-I../stored_apps/golf/src', '-DSTM32F429_439xx', '-DMAX_FONT_GLYPH_SIZE=256', '-DTARGET_QEMU', '-DIS_BIGBOARD', '-DDEFAULT_LOG_LEVEL=LOG_LEVEL_INFO', '-DPBL_LOG_ENABLED', '-DPBL_LOGS_HASHED', '-DBOOTLOADER_TEST_STAGE1=0', '-DBOOTLOADER_TEST_STAGE2=0', '-DPULSE_EVERYWHERE=1', '-DBT_CONTROLLER_QEMU', '-DRELEASE', 'stored_apps/golf/appinfo.auto.c', '-c', '-o/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/build/stored_apps/golf/appinfo.auto.c.4.o']
```

Thanks

Hexxeh, 2025-04-01:
Are you using the ARM toolchain distribution mentioned in the readme?

jibraniqbal666, 2025-04-01:
> Are you using the ARM toolchain distribution mentioned in the readme?

I downloaded this https://developer.arm.com/-/media/Files/downloads/gnu/14.2.rel1/binrel/arm-gnu-toolchain-14.2.rel1-darwin-arm64-aarch64-none-elf.pkg

jibraniqbal666, 2025-04-01:
I have remove the one which i downloaded from brew and got the binary instead.

```
❯ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/arm-gnu-toolchain/bin/../libexec/gcc/arm-none-eabi/14.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: /Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/src/gcc/configure --target=arm-none-eabi --prefix=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/build-arm-none-eabi/install --with-gmp=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/build-arm-none-eabi/host-tools --with-mpfr=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/build-arm-none-eabi/host-tools --with-mpc=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/build-arm-none-eabi/host-tools --with-isl=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/build-arm-none-eabi/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-gnu-as --with-headers=yes --with-gnu-ld --with-native-system-header-dir=/include --with-sysroot=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-14/build-arm-none-eabi/install/arm-none-eabi --with-bugurl=https://bugs.linaro.org/ --with-multilib-list=aprofile,rmprofile --with-pkgversion='Arm GNU Toolchain 14.2.Rel1 (Build arm-14.52)'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.1 20241119 (Arm GNU Toolchain 14.2.Rel1 (Build arm-14.52))
```

but i am getting different error now.

```
❯ ./waf configure --board=snowy_bb2 --qemu
Waf: The wscript in '/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp' is unreadable
Traceback (most recent call last):
  File "/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Scripting.py", line 105, in waf_entry_point
    set_main_module(wscript)
    ~~~~~~~~~~~~~~~^^^^^^^^^
  File "/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Scripting.py", line 145, in set_main_module
    Context.g_module=Context.load_module(file_path)
                     ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Context.py", line 364, in load_module
    exec(compile(code,path,'exec'),module.__dict__)
    ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble-esp/wscript", line 5, in <module>
    import pexpect
ModuleNotFoundError: No module named 'pexpect'
```

Hexxeh, 2025-04-01:
Do you have the virtualenv setup and requirements installed?

jibraniqbal666, 2025-04-02:
Hey it works I had to reinstall the pip requirements for some reason. Thanks.

I am now having issue installing qemu https://github.com/pebble-dev/qemu, but i believe that is irrelevant.

closed 2025-04-02 by jibraniqbal666

jibraniqbal666, 2025-04-03:
How did I fix it for future reference is to download the correct arm binary and not use the brew one and link it to the /usr/bin/local and delete all the rest conflicting ones, and don't forget to enable python env.

https://developer.arm.com/-/media/Files/downloads/gnu/14.2.rel1/binrel/arm-gnu-toolchain-14.2.rel1-darwin-arm64-arm-none-eabi.tar.xz

speccy88, 2025-04-30:
I'm stuck with the same issue, (stdint.h, no such file)
I re-run the pip command to install the requirements but everything is installed fine. Any ideas?

jibraniqbal666, 2025-06-01:
did you download and move the binary to /usr/bin/local?
then do `source .venv/bin/activate`