pebble-dev/pebble-firmware #130: Nimble does not use OS-provided root keys

Hexxeh, 2025-04-01:
We're using https://github.com/pebble-dev/pebble-firmware/blob/main/src/include/bluetooth/init.h#L27 at the moment and Nimble is generating its own, we should fix that.

Hexxeh, 2025-04-01:
Actually it looks like the issue is that we don't implement `ble_hs_cfg.store_gen_key_cb` and so our IRK is using the default baked into Nimble (`ble_hs_pvcy_default_irk`) for everyone and LTKs are just randomly generated.

We should pass through the host provided keys and use those to implement `store_gen_key_cb`.

gmarull, 2025-04-09:
I think it's ok to use random LTK (default NimBLE behavior). It looks like once bonding is stored, both main/PRF can reconnect just fine, no matter who stored it. Randomly generated IRK support was added in https://github.com/pebble-dev/pebble-firmware/pull/148 (even though watch does not use RPA for now).

closed 2025-04-09 by Hexxeh