pebble-dev/pebble-firmware #136: Building Qemu for emulation issue on M1 pro

jibraniqbal666, 2025-04-03:
I am trying to build Qemu from [pebble-dev/qemu](https://github.com/pebble-dev/qemu), but unfortunately I am getting this error while building the qemu, If please someone can help.

```
./configure  --enable-debug --target-list="arm-softmmu" \
    --extra-cflags=-DSTM32_UART_NO_BAUD_DELAY
```

```
$ make

Mouse support     yes
  GEN   config-host.h
  GEN   trace/generated-tracers.h
  GEN   trace/generated-tcg-tracers.h
  GEN   trace/generated-helpers-wrappers.h
  GEN   trace/generated-helpers.h
  CC    ui/egl-helpers.o
In file included from ui/egl-helpers.c:11:
/Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble_qemu/include/ui/egl-helpers.h:5:10: fatal error: 'epoxy/egl.h' file not found
#include <epoxy/egl.h>
         ^~~~~~~~~~~~~
1 error generated.
make: *** [ui/egl-helpers.o] Error 1
```

I found that you were able to figure it out, but i can't, please If you can?

Thanks

jibraniqbal666, 2025-04-03:
> I am trying to build Qemu from [pebble-dev/qemu](https://github.com/pebble-dev/qemu), but unfortunately I am getting this error while building the qemu, If please someone can help.
>
> ```
> ./configure [...]

Just had to disable gtk and opengl

```
./configure --disable-werror --enable-debug --target-list="arm-softmmu"  --disable-opengl --disable-gtk \
    --extra-cflags=-DSTM32_UART_NO_BAUD_DELAY --python=/usr/local/bin/python2
```

I don't know the side effect but it is posting for now, so i am happy.

[image]

DX37, 2025-04-19:
> > I am trying to build Qemu from [pebble-dev/qemu](https://github.com/pebble-dev/qemu), but unfortunately I am getting this error while building the qemu, If please someone can help.
> > ```
> > ./configure  --enable-debug --target-list="arm-softmmu" \
> >     --extra-cflags=-DSTM32_UART_NO_BAUD_DELAY
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ```
> > $ make
> > 
> > Mouse support     yes
> >   GEN   config-host.h
> >   GEN   trace/generated-tracers.h
> >   GEN   trace/generated-tcg-tracers.h
> >   GEN   trace/generated-helpers-wrappers.h
> >   GEN   trace/generated-helpers.h
> >   CC    ui/egl-helpers.o
> > In file included from ui/egl-helpers.c:11:
> > /Users/jibraniqbal/AndroidStudioProjects/pebble-next/pebble_qemu/include/ui/egl-helpers.h:5:10: fatal error: 'epoxy/egl.h' file not found
> > #include <epoxy/egl.h>
> >          ^~~~~~~~~~~~~
> > 1 error generated.
> > make: *** [ui/egl-helpers.o] Error 1
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I found that you were able to figure it out, but i can't, please If you can?
> > Thanks
> 
> Just had to disable gtk and opengl
> 
> ```
> ./configure --disable-werror --enable-debug --target-list="arm-softmmu"  --disable-opengl --disable-gtk \
>     --extra-cflags=-DSTM32_UART_NO_BAUD_DELAY --python=/usr/local/bin/python2
> ```
> 
> I don't know the side effect but it is posting for now, so i am happy.
> <img alt="Image" width="170" src="https://private-user-images.githubusercontent.com/30576574/430151553-dd0ac204-54b7-49e8-8acc-39278a2d0673.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDUwNjA5MzQsIm5iZiI6MTc0NTA2MDYzNCwicGF0aCI6Ii8zMDU3NjU3NC80MzAxNTE1NTMtZGQwYWMyMDQtNTRiNy00OWU4LThhY2MtMzkyNzhhMmQwNjczLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDE5VDExMDM1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJiMDI1OGJhMDYxYTNjYmRhMjkzZjI1MzgxNDY0NjY5YzczOGRhMjI5ZDA2Y2NmNmFmMjhmZDQ0ZmJlMmMxYjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.YguuwtmkyNSFB8sf64JD5YpGXKWRgmTjDPVQUBJWdi4">

Which compiler you used? And what was it's version?

DX37, 2025-04-29:
> Which compiler you used? And what was it's version?

Nevermind, had to use exactly Ubuntu 24.04 LTS. I was using Arch Linux, btw.