pebble-dev/pebble-firmware #187: Cannot translate strings

DX37, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Hello there! Trying to make first contributions here, but these errors occurred before me when I'm tried to pack some languages (ru_RU or even de_DE) below.
Disclaimer: I'm using Arch Linux and figured how to build QEMU on it and it run `snowy` and `silk` pretty good.
```
# ./waf --lang ru_RU pack_lang 
Waf: Entering directory `/home/dx37/pebble-firmware/build'
Warning: This PO file contains untranslated strings! 
/home/dx37/pebble-firmware/resources/normal/base/lang/ru_RU/tintin.po:2: warning: header field 'PO-Revision-Date' missing in header
/home/dx37/pebble-firmware/resources/normal/base/lang/ru_RU/tintin.po:2: warning: header field 'Last-Translator' missing in header
/home/dx37/pebble-firmware/resources/normal/base/lang/ru_RU/tintin.po:2: warning: header field 'Language-Team' missing in header
0 translated messages.
Created mo at /home/dx37/pebble-firmware/build/resources/normal/base/lang/ru_RU/tintin.po.STRINGS.mo 
Traceback (most recent call last):
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Scripting.py", line 122, in waf_entry_point
    run_commands()
    ~~~~~~~~~~~~^^
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Scripting.py", line 185, in run_commands
    ctx=run_command(cmd_name)
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Scripting.py", line 176, in run_command
    ctx.execute()
    ~~~~~~~~~~~^^
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Scripting.py", line 386, in execute
    return execute_method(self)
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Build.py", line 93, in execute
    self.execute_build()
    ~~~~~~~~~~~~~~~~~~^^
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Build.py", line 96, in execute_build
    self.recurse([self.run_dir])
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Context.py", line 133, in recurse
    user_function(self)
    ~~~~~~~~~~~~~^^^^^^
  File "/home/dx37/pebble-firmware/wscript", line 1722, in pack_lang
    ctx.recurse('resources/normal/base/lang')
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dx37/pebble-firmware/.waf3-2.1.4-72787ce48f227ac42c4b0da24e780694/waflib/Context.py", line 133, in recurse
    user_function(self)
    ~~~~~~~~~~~~~^^^^^^
  File "/home/dx37/pebble-firmware/resources/normal/base/lang/wscript", line 53, in pack_lang
    pack_lang_code(ctx, lang_code)
    ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/dx37/pebble-firmware/resources/normal/base/lang/wscript", line 89, in pack_lang_code
    resource_data[name] = f.read()
                          ~~~~~~^^
  File "<frozen codecs>", line 325, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xde in position 0: invalid continuation byte
```

Also, maybe because of inability of packing any language, in PebbleOS there's only English available.

[image]

Is this because of my too fresh Linux installation and I still should use Ubuntu 24.04, or it's something new?

### Expected Behavior

Ability to translate language strings and pack them to test at least on PebbleOS in QEMU.

### Steps To Reproduce

1. Configure repo just with Getting Started docs
2. Try to pack any non `en_US` language

### Environment

```markdown
- **OS**: Arch Linux
- **Watch**: not relevant
```

DX37, 2025-05-02:
Okay, I think I resolved this error with changing file read mode to binary in `resources/normal/base/lang/wscript`
`with open(mo_path, 'rb') as f:`
But I guess resources not flashing into QEMU emulation. Anyway, lang is now packing, there's still one SyntaxWarning though:
```
./waf --lang ru_RU pack_lang             
Waf: Entering directory `/home/dx37/pebble-firmware/build'
Warning: This PO file contains untranslated strings! 
/home/dx37/pebble-firmware/resources/normal/base/lang/ru_RU/tintin.po:2: warning: header field 'PO-Revision-Date' missing in header
/home/dx37/pebble-firmware/resources/normal/base/lang/ru_RU/tintin.po:2: warning: header field 'Last-Translator' missing in header
/home/dx37/pebble-firmware/resources/normal/base/lang/ru_RU/tintin.po:2: warning: header field 'Language-Team' missing in header
3 translated messages, 2 fuzzy translations, 567 untranslated messages.
Created mo at /home/dx37/pebble-firmware/build/resources/normal/base/lang/ru_RU/tintin.po.STRINGS.mo 
/home/dx37/pebble-firmware/tools/generate_codepoint_requirements.py:26: SyntaxWarning: invalid escape sequence '\w'
  langstr = re.search("^\"Language: (\w)", line)
Building empty resource BITHAM_18_LIGHT_SUBSET_EXTENDED 
Building empty resource BITHAM_30_BLACK_EXTENDED 
Building empty resource BITHAM_34_LIGHT_SUBSET_EXTENDED 
Building empty resource BITHAM_34_MEDIUM_NUMBERS_EXTENDED 
Building empty resource BITHAM_42_BOLD_EXTENDED 
Building empty resource BITHAM_42_LIGHT_EXTENDED 
Building empty resource BITHAM_42_MEDIUM_NUMBERS_EXTENDED 
Building empty resource ROBOTO_CONDENSED_21_EXTENDED 
Building empty resource ROBOTO_BOLD_SUBSET_49_EXTENDED 
Building empty resource DROID_SERIF_28_BOLD_EXTENDED 
Created language pack at /home/dx37/pebble-firmware/build/resources/normal/base/lang/ru_RU/ru_RU.pbl 
Waf: Leaving directory `/home/dx37/pebble-firmware/build'
'pack_lang' finished successfully (0.137s)
```

DX37, 2025-05-02:
Warning is gone after I added `r` to `langstr = re.search(r"^\"Language: (\w*)", line)` string in `tools/generate_codepoint_requirements.py`

DX37, 2025-05-02:
Any idea how to add resources, especially language pack, into QEMU emulation?

hellcp, 2025-05-02:
PBLs are usually streamed by the mobile app, and there only exist one language pack on the watch at any given time. There's the pbl tool that can handle installing language packs though https://github.com/pebble-dev/pebble-firmware/blob/main/python_libs/pbl/pbl/commands/install_lang.py

DX37, 2025-05-02:
Don't understand how to use it, stuck with not working `pebble-commander` for some irrelevant reason.

PR #192 (2025-05-05): i18n: fix languages packing
fix `./waf` commands `pack_lang` and `pack_all_langs`
fixes #187
commit 3d5193fd (2025-05-05): i18n: fix languages packing

fix `./waf` commands `pack_lang` and `pack_all_langs`
fixes #187
closed 2025-05-05 by gmarull