pebble-dev/pebble-firmware #242: TYPICAL text doesn't account for different forms depending on different days of the week

hellcp, 2025-05-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

When translating `TYPICAL %s` string from `../src/fw/apps/system_apps/health/health_ui.c:66`, I noticed that this string might not be conjugated correctly. In polish the weekdays have two different gender forms:
- masc: poniedziałek (monday), wtorek (tuesday), czwartek (thursday), piątek (friday)
- fem: środa (wednesday), sobota (saturday), niedziela (sunday)
and depending on the gender, the TYPICAL in that sentence would be translated as:
- masc: TYPOWY (ex. TYPOWY PONIEDZIAŁEK)
- fem: TYPOWA (ex. TYPOWA ŚRODA)

### Expected Behavior

It is possible to translate with the correct conjugation somehow. That could be done by combining the word typical with the 7 possible options.

### Steps To Reproduce

N/A

### Environment

```markdown
N/A
```

Vrabbers, 2025-05-19:
This is also a problem for Portuguese.

Típic**a** segunda/terça/quarta/quinta/sexta-feira for Monday-Friday, but Típic**o** sábado/domingo for Saturday and Sunday.

welaq, 2025-05-19:
There is also a problem with the only one "May" string. In English it used as abbreviation (Jan, Feb, Mar, Apr, **May**...) as well as full name of the month (January, February, March, April, **May.**..). Meanwhile in some other languages we need two different strings.

DX37, 2025-05-20:
Russian have this as well, but even more difficult:

* Типичн**ый** - Понедельник, Вторник, Четверг (male)
* Типичн**ая** - Среда, Пятница, Суббота (female)
* Типичн**ое** - Воскресенье (neuter)

welaq, 2025-05-21:
In Crowdin one user noted that in string "TYPICAL %s" variable stands for abbreviated weekday (3 letters).
So the final string is going to be **TYPICAL MON** or **TYPICAL TUE** etc...
Also you can see it in this [video](https://youtu.be/KQh1b_srGM4?si=ySOT93aTD1xQyECQ&t=84)
However that video is old and I am not sure has anything changed in the source code since then.