pebble-dev/pebble-firmware #245: CC2564 based devices exhaust ACL buffers during periods of heavy PPoGATT traffic

Hexxeh, 2025-05-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Eventually after several attempts, the watch will assert as a result of this check failing: https://github.com/apache/mynewt-nimble/blob/9d0217ac673c7912df98608f1f53ca6479515285/nimble/transport/common/hci_h4/src/hci_h4.c#L135 

### Expected Behavior

Installing large apps repeatedly should not crash the watch.

### Steps To Reproduce

Install an app such as PinyWings repeatedly, either clearing the app after each install or cancelling it before it manages to complete.

### Environment

```markdown
- **Watch**: snowy_bb2
```

jwise, 2025-07-27:
I was previously able to reproduce this even after having added my own version of this patch.  Then, I added debugging code inside NimBLE to instrument buffer alloc and dealloc, and I could no longer reproduce this anymore!  What gives?

jwise, 2025-07-27:
Note that this can be distinguished from a normal sync loss assert with something like:

```
@@ -132,6 +135,7 @@ hci_h4_sm_w4_header(struct hci_h4_sm *h4sm, struct hci_h4_input_buffer *ib)
         assert(h4sm->allocs && h4sm->allocs->acl);
         h4sm->om = h4sm->allocs->acl();
         if (!h4sm->om) {
+            PBL_LOG(LOG_LEVEL_ERROR, "HCI_H4_ACL buf alloc");
             return -1;
         }
```

in `third_party/nimble/mynewt-nimble/nimble/transport/common/hci_h4/src/hci_h4.c`.

PR #264 (2025-07-27): nimble/cc2564x: turn on flow control for HCI
Enables flow control for HCI, and makes as surgical of restructuring of the interrupt handler as possible such that deallocation happens on the high-priority NimBLE UART thread, rather than on the main thread.

Fixes #245 (I hope).  With this fix, I spent most of an evening banging on it trying to get it to blow up, without tripping it at all (though last night I did see it regularly, which makes me think I must have forgotten to recompile before I flashed my device, or something).  Without this fix, I crash within two connection attempts with the ` [... cut]