pebble-dev/pebble-firmware #261: instability on snowy_bb2 around cycling airplane mode

jwise, 2025-07-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Varying instability cycling airplane mode on snowy_bb2.

I have seen a MemManage fault (where something tries to dereference `0xbdbdbdbd`) but cannot reproduce it now.  This one has:

```
02:33:24.193 E S 02:33:23.511 assert.c:21> assert at line 302, func: ble_svc_gap_init - rc == 0
```

GDB has a stack trace of:

```
#5  0x0802e3f8 in ble_svc_gap_init () at ../third_party/nimble/mynewt-nimble/nimble/host/services/gap/src/ble_svc_gap.c:302
#6  0x0802d882 in bt_driver_start (config=config@entry=0x10003410) at ../src/bluetooth-fw/nimble/init.c:115
#7  0x08047d40 in prv_comm_start () at ../src/fw/services/common/bluetooth/bluetooth_ctl.c:93
#8  0x08047e3e in prv_comm_start () at ../src/fw/services/common/bluetooth/bluetooth_ctl.c:174
#9  prv_comm_state_change (context=<optimized out>) at ../src/fw/services/common/bluetooth/bluetooth_ctl.c:160
#10 0x0804f2c4 in system_task_main (paramater=<optimized out>) at ../src/fw/services/common/system_task.c:79
#11 0x00000000 in ?? ()
```

which is a failure on the `ble_gatts_add_svcs` call.  Unfortunately, by this point, `rc` is long since `<optimized out>`, but it seems like `ble_hs_conn_first` might have returned something, because `ble_hs_conns` does contain a connection, which it shouldn't at that point during init.

### Expected Behavior

no crash on cycling airplane mode

### Steps To Reproduce

cycle airplane mode sometimes on snowy_bb2

### Environment

```markdown
- **OS**: iOS 18.5 (22F76) on iPhone 15 Pro
- **Watch**: snowy_bb2
- **Mobile app**: 4.4
```

jwise, 2025-07-27:
`0xBD` comes from the heap poisoning in `heap_free`.

Hexxeh, 2025-07-27:
`ble_hs_conn_first` is returning a connection even though the stack stop supposedly succeeded and the callback for `ble_hs_stop` got called with success, which is why `ble_gatts_mutable` is returning false and preventing `ble_gatts_reset` from running.

Hexxeh, 2025-07-29:
PebbleOS terminates all GAP connections before tearing the stack down, but as part of stack teardown Nimble also tries to terminate for connections, and then wait for them to terminate. Because the termination is already in progress, Nimble logs and error, doesn't wait for the termination to complete and erroneously claims success anyway.

I was going to fix this, but I happened to notice it's actually fixed upstream via https://github.com/coredevices/mynewt-nimble/commit/b9035d28c68a19789a1427a029b03fa759d906f9 so I'll just update our Nimble fork to pull in the latest changes including this one.

closed 2025-07-29 by Hexxeh